OceanographyforEveryone/OpenCTD #11: Qduino mini board manager

camullen86, 2017-04-18:
When preparing a sketch using the qduino mini i noticed the IDE does not have drivers for this board and found info online suggesting it can easily be bricked if using the wrong setup. i was able to find the board manager and set it up easily enough but a note reminding makers to be aware of this may be good.

i used the following site to get mine set as it has full directions.
https://www.hackster.io/team-qtechknow/qduino-mini-quickstart-guide-8b2d68

SouthernFriedScientist, 2017-04-18:
Thank you! I added the Qtechknow quickstart guide to the README.

closed 2017-04-18 by SouthernFriedScientist

camullen86, 2017-04-23:
Hi Andrew,
i was wondering if you had any issues getting the Qduino Mini to work with
the IDE. I've gone through the full setup to install the board manager and
drivers, everything is up to date. When i launch the IDE it shows Qduino
Mini on COM PORT2 in the bottom right of the application, and is also
selected as the board being used, the problem I'm having is that the Ports
menu is grayed out and inaccessible, and if i try to compile and upload
anything it says it cant reach the board.

there appears to be a hardware issue that says the device needs further
installation, but the drivers and files are all up to date.
So far I haven't been able to find anything useful in any of the forums
about this subject.

Thanks for any insight you may have,
Chris

SouthernFriedScientist, 2017-05-09:
Hi Chris,

If the comm ports don't show up, that generally means there's a driver issue. Have you tried installing all the default Arduino drivers, not just the Qduino specific ones?

If that doesn't work, try pinging https://twitter.com/Qtechknow, they are super helpful and responsive.