OceanographyforEveryone/OpenCTD #15: Inquiry about Arduino UNO with CTD

hpollek, 2017-06-07:
Hello, I am pretty new to Arduino and am trying to build the open CTD 
I was wondering if anyone had suggestions for a pinout guide for use with the Arduino UNO to avoid the failed conductivity readings mentioned with the use of the given pinout on the UNO
Thanks

CTDizzle, 2017-06-07:
hpollek,

I managed to get my first version of the OpenCTD running on the Uno, but
unfortunately I can't seem to find the old code.
I do recall that I was able to get it to read the conductivity, but with
some gibberish. (Now that I think about it, that was probably my fault and
not a function of the UNO)

I think it is definitely possible to get it to work reliably on the UNO.
Atlas-Scientific actually used it when they wrote their sample code
<https://www.atlas-scientific.com/_files/code/Arduino-Uno-EC-sample-code.pdf>.
Try running through their example using the pinouts they use. I think they
define (rx,tx) as (2,3). Once you get that working, you can redefine those
pinouts to suit your needs.

One thing that Atlas-Scientific mentions in their datasheet is to try to
look at the readings while the probe is in solution. If you are getting bad
data in open air, try placing the probe in a cup of tap water. If your data
then looks reasonable, then you can assume that your probe is creating some
interference. Here is the EC EZO datasheet
<https://www.atlas-scientific.com/_files/_datasheets/_circuit/EC_EZO_Datasheet.pdf>.
Page 12 will give you a better description.

Please let me know if you need anymore help. I am still an Arduino amateur
and I could use the practice explaining how things work.

Cheers,
- Ian

-- 
*Ian Black*
M.S. Candidate, Marine Resource Management
Ocean Observatories Initiative (OOI) Technician
College of Earth, Ocean, and Atmospheric Sciences
Oregon State University
Office: 312 Strand Agriculture Hall
Cell: (541) 817-3687

*Interested in OSU's involvement with the OOI?*
Information about the Endurance Array
<http://oceanobservatories.org/array/coastal-endurance/>
Information about Global Station Papa
<http://oceanobservatories.org/array/global-station-papa/>

*More Interesting Things*
Live Feed at Axial Seamount
<http://oceanobservatories.org/streaming-underwater-video/>

hpollek, 2017-06-07:
Thanks a bunch, I got the conductivity sensor to work in the serial monitor!
now the only issue is with the SD card, because at the top of the serial monitor window it says card failed.
Any advice on the SD card pinouts would also be greatly appreciated

CTDizzle, 2017-06-08:
What brand and type of SD card module are you using?

If you are using something similar to what is used in the guide, you'll want to connect the SCK, MOSI, and MISO pins to the same defined pins on the UNO. 

I think those pins are:

SCK - Pin 13
MOSI - Pin 11
MISO - Pin 12
VCC - 3V3
GND - GND

I think the CS pin on the SD module can be connected to any pin you define. The SD library appears to default to pin 10 for CS. If you look at the OpenCTD_qDuino code, you'll notice the lines...

`if (!SD.begin(4))`
`Serial.println("Card failed");  // Log to serial monitor if SD card doesn't work.`

This basically means "if no SD card is detected on pin 4 (the CS), then display "card failed" to the serial monitor." So try connecting your CS to whatever pin you want and place that pin number in those brackets. 

The reference page for [ Serial Peripheral Interface (SPI)](https://www.arduino.cc/en/Reference/SPI) provides a good explanation of the how the protocol works. SPI is the protocol used to communicate with the SD module.

hpollek, 2017-06-08:
Thanks for the help, 
my SD card reader looks like this and is from LC technology
[image]
I have my pins attached as you mentioned with the corresponding number for the CS in the Sketch, but was still getting "card failed" messages and wondering what the number in PinMode() just above the if statement you mentioned should be?

CTDizzle, 2017-06-08:
I see that you have your VCC hooked up to 5v. Try hooking it up to 3.3v.
Make sure it is also hooked up to 3.3v on the UNO.
If that doesn't work, also try changing the SD.begin(10)

I think I have a similar SD card module at home. Let me try a similar set
up and I will try to get back to you in a few hours if I get some
reasonable results.

-- 
*Ian Black*
M.S. Candidate, Marine Resource Management
Ocean Observatories Initiative (OOI) Technician
College of Earth, Ocean, and Atmospheric Sciences
Oregon State University
Office: 312 Strand Agriculture Hall
Cell: (541) 817-3687

*Interested in OSU's involvement with the OOI?*
Information about the Endurance Array
<http://oceanobservatories.org/array/coastal-endurance/>
Information about Global Station Papa
<http://oceanobservatories.org/array/global-station-papa/>

*More Interesting Things*
Live Feed at Axial Seamount
<http://oceanobservatories.org/streaming-underwater-video/>

CTDizzle, 2017-06-09:
I hooked up an old UNO to a DS18B20 temperature sensor and an SD card module. The code I used is below. You can actually exclude the pinMode line.

 You may also try to format your SD card if you haven't already.
- Plug your SD card into your laptop.
- In Windows Explorer, right click on the SD card. Select Format.
- Under File System, select FAT.
- Select Start. 

Pinouts used for the SD Module to UNO.
- GND to GND
- 3.3v to 3.3v
- CS to 4
- MOSI to 11
- SCK to 13
- MISO to 12
Don't hook the 5v to anything. 

If you run the code below (with just your temperature sensor hooked up), you should see the temperature reported to the serial monitor every second. Keep in mind that I only used one temperature sensor, so you may need to adapt the code below to work for you. Then once you pull the SD card, you should see a file named "datalog.txt" that contains the same temperatures you saw in the serial monitor.

`#include <Wire.h>   //Native in Arduino IDE.              
#include <OneWire.h> //Third party library.
#include <DallasTemperature.h>  //Third party library.
#include <SPI.h>   //Native in Arduino IDE.              
#include <SD.h>    //Native in Arduino IDE.              

float temp;   //Make the temperature value a floating point number.
OneWire oneWire(6);   //Data line on temperature sensor goes to pin 6 of arduino.
DallasTemperature tempsensor(&oneWire); //Call DallasTemperature function to "tempsensor"

void setup() {  //Start your engines.
  Serial.begin(9600);   //Set the baud rate to 9600 bps.
  if (SD.begin(4)){   //If an SD card IS detected (with CS to pin 4).
    tempsensor.begin();   //Initialize the temperature sensor.
    delay (500);    //Wait half a second.
  }
  else{
    Serial.println("No card detected. Check to see if card is properly installed."); //If no card is detected, print this message.
    delay(10);  //Wait 10 milliseconds.
    return;
  }
  delay(500);   //Wait half a second.
}

void loop() { //And around we go.
  tempsensor.requestTemperatures(); //Request the temperature from the DS18B20.
  temp =tempsensor.getTempCByIndex(0);  //Define "temp" as the degC value from the probe.
  
  File dataFile = SD.open("datalog.txt", FILE_WRITE); //Create or open a file on the SD card titled "datalog.txt".
  if (dataFile) {   //If dataFile is available,
    dataFile.println(temp); //Print the temperature to it.
    dataFile.close();   //Close the file.

    Serial.println(temp); //Then print that temperature to the serial monitor if the file opened.
  }
  delay(1000);  //Repeat the loop every second.
}`

SouthernFriedScientist, 2017-06-12:
Just wanted to jump in to say that we've found about 5% of those SD card reader/writers are defective out of the package (it's always hard to tell who really makes them because the suppliers/branding change alot, even though they tend to be made by the same factory), so it might also be worth testing the SD reader alone with a known piece of code just to make sure it's not the hardware.

closed 2019-05-20 by SouthernFriedScientist