OceanographyforEveryone/OpenCTD #16: Any way to make the CTD compatible with M0 Pro?

hpollek, 2017-06-09:
Hello, after getting the CTD working fully on my UNO I now have yet another question regarding making the Open CTD work with another board, and this one is giving me more grief.

I'm aiming to potentially get better resolution on the readings by adapting the CTD to be used with the M0 Pro, I have all of the hardware working on the UNO, so now where I've hit this speed bump is with the Sketch, and more specifically, the SoftwareSerial library.

The M0 Pro board does not have SoftwareSerial as a built-in library, as far as I have found online thus far, it is only usable with other Serial functions. I am not very experienced in the wide range of different Arduino board but figured this may be a relatively obvious work-around for someone with more experience and I would appreciate any help that I can get.

Is there any way of achieving the same results as SoftwareSerial with a different approach that would work on this board?
Thanks!

CTDizzle, 2017-06-10:
hpollek,

I have run into the same issue when trying to use the Arduino MKRZero. I
think there is a work around using SamdSercom (
https://www.arduino.cc/en/Tutorial/SamdSercom). You might be able to find a
solution if you look into that and code samples you find around the
internet.
I also haven't tried it out, but maybe you could set the EC EZO to I2C mode
to bypass the need for a Rx/Tx and SoftwareSerial? Atlas-Scientific offers
a tutorial and some sample code for that.

I haven't had the time to really sit down and work out a solution. I have a
research cruise in July that will give me a few days of downtime, so maybe
I will work on it then.

If you do come up with a solution, please let me know because I would
really like to start using the MKR board series for the CTD.

Cheers,
- Ian

<https://github.com/OceanographyforEveryone/OpenCTD/issues/16>, or mute the
thread

CTDizzle, 2017-06-30:
Any luck finding a solution?

hpollek, 2017-06-30:
Initially the Sketch wouldn't even upload to the m0 pro
Now the Sketch is to the point that it compiles and uploads, but the serial monitor shows no readings

CTDizzle, 2017-06-30:
Mind if I take a look at your sketch? You can post it here or email it to blackia@oregonstate.edu
- Ian

SouthernFriedScientist, 2019-05-20:
Hey folks, coming back late to this, but we now have a SoftwareSerial library that will work with M0 boards. You can find it here: https://github.com/OceanographyforEveryone/OpenCTD/tree/master/OpenCTD_Feather_Adalogger/Support_Code/SoftwareSerial

closed 2019-05-20 by SouthernFriedScientist