OceanographyforEveryone/OpenCTD #20: MS5803 Pressure sensor; Software issue

Jake-DeCicco, 2018-07-31:
**Essentially, the sensor reports data to the serial monitor when I upload the following code (sorry about the formatting):**

#include <Wire.h>
#include <SparkFun_MS5803_I2C.h>

//  ADDRESS_HIGH = 0x76
//  ADDRESS_LOW  = 0x77

MS5803 sensor(ADDRESS_HIGH);

float temperature_c, temperature_f;
double pressure_abs, pressure_relative, altitude_delta, pressure_baseline;

double base_altitude = 1655.0; // Altitude of SparkFun's HQ in Boulder, CO. in (m)

void setup() {
    Serial.begin(9600);
    sensor.reset();
    sensor.begin();
}

void loop() {
  
  temperature_c = sensor.getTemperature(CELSIUS, ADC_512);

  pressure_baseline = sensor.getPressure(ADC_4096);

  temperature_f = sensor.getTemperature(FAHRENHEIT, ADC_512);
  
  pressure_abs = sensor.getPressure(ADC_4096);

  pressure_relative = sealevel(pressure_abs, base_altitude);

  altitude_delta = altitude(pressure_abs , pressure_baseline);
  
  // Report values via UART
  Serial.print("Temperature C = ");
  Serial.println(temperature_c);
  
  Serial.print("Temperature F = ");
  Serial.println(temperature_f);
  
  Serial.print("Pressure abs (mbar)= ");
  Serial.println(pressure_abs);
   
  Serial.print("Pressure relative (mbar)= ");
  Serial.println(pressure_relative); 
  
  Serial.print("Altitude change (m) = ");
  Serial.println(altitude_delta); 

  delay(1000);

  }
  
 double sealevel(double P, double A)
// Given a pressure P (mbar) taken at a specific altitude (meters),
// return the equivalent pressure (mbar) at sea level.
// This produces pressure readings that can be used for weather measurements.
{
	return(P/pow(1-(A/44330.0),5.255));
}

double altitude(double P, double P0)
// Given a pressure measurement P (mbar) and the pressure at a baseline P0 (mbar),
// return altitude (meters) above baseline.
{
	return(44330.0*(1-pow(P/P0,1/5.255)));
}
  

**But fails to do so when I upload code for the entire function of the CTD, even when I eliminate the other sensors from the active code.**

#include <Wire.h>               
#include <SparkFun_MS5803_I2C.h>
#include <MS5803_14.h>
#include <SoftwareSerial.h>
#include <OneWire.h>        
#include <DallasTemperature.h>
#include <SPI.h>               
#include <SD.h>                

float tempA;
float tempB;
float tempC;

float tempAvg;

float EC_float = 0;  // Electrical conductivity.

char EC_data[48];    // A 48 byte character array to hold incoming data from the conductivity circuit. 
char *EC;            // Character pointer for string parsing.

byte received_from_sensor = 0;  // How many characters have been received.
byte string_received = 0;       // Whether it received a string from the EC circuit.

SoftwareSerial ecSerial(8, 9);      // Define the SoftwareSerial port for conductivity.
OneWire oneWire(6);        // Define the OneWire port for temperature.
DallasTemperature sensors(&oneWire);  // Define DallasTemperature input based on OneWire.
MS5803 sensor(ADDRESS_HIGH); //Define pressure sensor

float temperature_c, temperature_f;
double pressure_abs, pressure_relative, altitude_delta, pressure_baseline;
double base_altitude = 1655.0;

void setup(void) {

  Serial.begin(9600);
  ecSerial.begin(9600); 

  pinMode(16, OUTPUT);  // Set data output pin for the SD card reader.
  
  if (!SD.begin(4)) {
    Serial.println("Card failed");
    return; 
  }
  delay(500);

  sensor.reset();
  sensor.begin(); 
  sensors.begin();  
  
  delay(250); 

}

void loop(void) {
  /*if (ecSerial.available() > 0) {
    received_from_sensor = ecSerial.readBytesUntil(13, EC_data, 48);

    // Null terminate the data by setting the value after the final character to 0.
    EC_data[received_from_sensor] = 0;
  }

  // Parse data, if EC_data begins with a digit, not a letter (testing ASCII values).
  if ((EC_data[0] >= 48) && (EC_data[0] <=57)) {
    parse_data();
  }
  delay(10);*/

  temperature_c = sensor.getTemperature(CELSIUS, ADC_512);
  temperature_f = sensor.getTemperature(FAHRENHEIT, ADC_512);
  pressure_abs = sensor.getPressure(ADC_4096);
  pressure_baseline = sensor.getPressure(ADC_4096);
  altitude_delta = altitude(pressure_abs , pressure_baseline);
  
  /*sensors.requestTemperatures();
  tempA = sensors.getTempCByIndex(0);
  tempB = sensors.getTempCByIndex(1);
  tempC = sensors.getTempCByIndex(2);
  tempAvg = (tempA + tempB + tempC) / 3;*/

  logData();  
  sendToSerialMonitor();
  delay(50);
}

void parse_data() {
  EC = strtok(EC_data, ",");                  
}

void logData() {
  /*File dataFile = SD.open("datalog.txt", FILE_WRITE);
  if (dataFile) {
    dataFile.print("Temperature C = ");
    dataFile.println(temperature_c);
    dataFile.print("Temperature F = ");
    dataFile.println(temperature_f);
    dataFile.print("Pressure abs (mbar)= ");
    dataFile.println(pressure_abs);
    dataFile.print("Pressure relative (mbar)= ");
    dataFile.println(pressure_relative); 
    dataFile.print("Altitude change (m) = ");
    dataFile.println(altitude_delta);
    dataFile.print("TempA = ");
    dataFile.print(tempA);
    dataFile.print("  TempB = ");
    dataFile.print(tempB);
    dataFile.print("  TempC = ");
    dataFile.print(tempC);
    dataFile.println("  ");
    dataFile.print("EC = ");
    dataFile.print(EC);
    dataFile.close();
 }
  else {

 }*/
}

void sendToSerialMonitor() {
  Serial.print("Temperature C = ");
  Serial.println(temperature_c);
  
  Serial.print("Temperature F = ");
  Serial.println(temperature_f);
  
  Serial.print("Pressure abs (mbar)= ");
  Serial.println(pressure_abs);
   
  Serial.print("Pressure relative (mbar)= ");
  Serial.println(pressure_relative); 
  
  Serial.print("Altitude change (m) = ");
  Serial.println(altitude_delta);
  
  /*Serial.print("tempA = ");
  Serial.print(tempA);
  Serial.print("  tempB = ");
  Serial.print(tempB);
  Serial.print("  tempC = ");
  Serial.print(tempC);
  Serial.print("  Avg = ");
  Serial.print(tempAvg);
  Serial.println("  ");
  
  Serial.print(EC);
  Serial.println("");*/
}

double altitude(double P, double P0){
// Given a pressure measurement P (mbar) and the pressure at a baseline P0 (mbar),
// return altitude (meters) above baseline.
  return(44330.0*(1-pow(P/P0,1/5.255)));
}
**And btw, thanks for answering my other question in such a timely manner.**

CTDizzle, 2018-07-31:
Hi Jake,

Would you mind emailing me your complete sketch for the CTD and the
hardware that you are using? I have some old CTD parts lying around and I
will take a look during lunch today.

You can send it to blackia@oregonstate.edu.

Thanks,
Ian Black

Squidman32, 2019-06-12:
Did this get resolved?  I'm having a similar issue.  bread boarded everything, and it all worked great.  Potted the depth sensor and data only comes in as 0.00.  So I assumed I screwed it up when I potted it.  I ordered two new depth sensors, when  I bread board both out, all I get is 0.00 coming in.  I cut and pasted your current code (as of 6/12/19) and it still reads 0.00  all the other temp and conductivity work no problem.  any ideas?

SouthernFriedScientist, 2019-06-13:
Good morning, 

Do you have 10K pull up resistors between SCL, SDA, and VCC?

Squidman32, 2019-06-13:
HI Andrew, Thanks so much for the fast response.  I was using the Sparkfun pressure sensor  that didn't require the 10K resistors.  It was working great, I potted it, it stopped working.  At the time I guessed maybe I got epoxy or something somewhere it should go so I tried over again.

I ordered off the shelf pressure sensor set ups like those recommended on the OpenCTD git hub page.  When the new ones came, I bread boarded one to the Qduino and still got 0.00 for readings.  So just to double check, I tried a second sensor, same result. Then I figured maybe the Qduino had a problem, so I swapped it out and solder one of the new pressure sensors to it, and still 0.00.  I'm running 3.3 volts from the VCC on the Qduino.  I tried switching around the SDL and SDA, still 0.00.  I tried loading the pressure sensor testing sketch, same readings.  I cut and pasted all the code straight from the git hub page incase I had a bug in my sketch some where, same result.  The temp and conductivity works no problem, so I'm running out of ideas. It is the weirdest thing as it all worked at one point.  
[image]
[image]
[image]
[image]

SouthernFriedScientist, 2019-06-13:
That is weird. The pressure sensor should be totally fine. Even if you got epoxy on it, it should still read something. I haven't played witht he qDuino code in a while now, but I seem to remember some pressure breakout boards needed the SD card reader to engage before they started reading. Have you tried connecting up everything? From those pictures, it doesn't look like there's anything on the hardware side that should cause it to fail. 

Are you using the qDuino code of the M0 code? You should still be using the qDUino code.

Squidman32, 2019-06-13:
Andrew you are a genius!!!!!  That was it. I hooked up the SD card reader and I'm back in business! The original one had a wire disconnect on the SD reader when I squeezed it in the tube after potting, and that caused the issue.  Thank you so much for your help!

SouthernFriedScientist, 2019-06-14:
Thanks! Though I wouldn't sing my praises too much, after all, I'm the one who wrote the janky code in the first place.

closed 2019-06-14 by SouthernFriedScientist