OceanographyforEveryone/OpenCTD #21: Calibrate Conductivity probe

Squidman32, 2019-03-13:
Under 6.3 conductivity the link " have created a small program for accessing the serial monitor" is dead.  I assumed it was the "simple_serial_for_EC_calibration" in the support code folder.  But when I flashed that over, I only got a stream of 0.00, and got no repsonse when I entered c,0 or k,? or any of the other commands into the serial monitor send command.  Any suggestions?

IanTBlack, 2019-03-13:
Hello,

What microcontroller are you using?  And what conductivity probe are you
using?
I believe you are correct for the support code folder.

My first thought is that you should make sure that your baud is set
correctly for your serial monitor. Another thing to check is to see if you
have the correct line ending selected.
Another thing to check is if you have correctly assigned your RX and TX in
your sketch to what matches for your board.

If you are using the Qduino and talking to the EC EZO via Serial, this is
the code that I used.
https://github.com/IanTBlack/CTDizzle/blob/master/Legacy/Mk3/Qduino_EC_Cal.ino
Give this a shot. If it doesn't work, let me know.

If you are talking to the EC EZO via I2C, you can try modifying this code
to work for your setup:
https://github.com/IanTBlack/CTDizzle/blob/master/Mk5/IDE_EC_Cal.ino

Feel free to get in touch if you have any more questions or issues.
Cheers,
-------------
Ian Black
Faculty Research Assistant | Ocean Observing Center
College of Earth, Ocean, and Atmospheric Sciences | Oregon State University
Cell: (541) 817-3687

IanTBlack, 2019-03-13:
Hello,

What microcontroller are you using?  And what conductivity probe are you
using?
I believe you are correct for the support code folder.

My first thought is that you should make sure that your baud is set
correctly for your serial monitor. Another thing to check is to see if you
have the correct line ending selected.
Another thing to check is if you have correctly assigned your RX and TX in
your sketch to what matches for your board.

If you are using the Qduino and talking to the EC EZO via Serial, this is
the code that I used.
https://github.com/IanTBlack/CTDizzle/blob/master/Legacy/Mk3/Qduino_EC_Cal.ino
Give this a shot. If it doesn't work, let me know.

If you are talking to the EC EZO via I2C, you can try modifying this code
to work for your setup:
https://github.com/IanTBlack/CTDizzle/blob/master/Mk5/IDE_EC_Cal.ino

Feel free to get in touch if you have any more questions or issues.
Cheers
-------------
Ian Black
Faculty Research Assistant | Ocean Observing Center
College of Earth, Ocean, and Atmospheric Sciences | Oregon State University
Cell: (541) 817-3687

Squidman32, 2019-03-13:
Got it to work!!! thank you.
I should have looked harder at your notes.  I had to type in c,0 a few
times before it responded.  Thanks for your help.

Out of curiosity, have you tried tossing an Oxygen sensor on one of these?
I've been look at Pyro Science's little piccolo 2 and/or
firestingO2-SubPort.  Its a bit over my head as this is literally my first
arduino type project and I'm limping along on other people's code.  But it
sure would make this CTD that much cooler if it could do O2.

Lloyd
Lloyd Trueblood Ph.D.
Associate Professor and Chair
Biology Department

<https://lasierra.edu/>

Lloyd.Trueblood@lasierra.edu
Phone: (951) 785-2 <(951)+785-2105>521
Fax: (951) 785-2100

lasierra.edu  <https://brand.lasierra.edu/email-signatures/#>
4500 Riverwalk Parkway | Riverside, CA 92505

Confidentiality: This e-mail (including any attachments) may contain
confidential, proprietary and privileged information, and unauthorized
disclosure or use is prohibited. If you receive this email in error, please
notify the sender and delete this e-mail from your system.

closed 2019-06-13 by SouthernFriedScientist