OceanographyforEveryone/OpenCTD #22: SoftwareSerial

Squidman32, 2019-10-15:
HI Guys, I'm having a library problem, and I'm sure the answer is obvious, but I"m still in the early stages of the github/arduino learning curve.  I can't find the downloadable ZIP file of the modified SoftwareSerial library for the Adafruit feather CTD.  I can navigate to the code on the GitHub page and see it.  But only option I have is to try and cut and paste it, there is no zip/download.  Where can I find the downloadable zip to add the library  to Arduino IDE?

IanTBlack, 2019-10-15:
Are you using the Feather 32u4 or Feather M0? If using the M0, you don't
need SoftwareSerial.
If you want me to give me your components specs, I might be able to fire up
some old code and give you a solution.

Cheers,
-------------
Ian Black
Faculty Research Assistant | Ocean Observing Center
College of Earth, Ocean, and Atmospheric Sciences | Oregon State University
Cell: (541) 817-3687

SouthernFriedScientist, 2019-10-15:
Hello! GitHub is terrible about making simple downloads easy. To get everything in a nice package you either need to clone the OpenCTD library and then access the local version on your desktop, or download the entire OpenCTD repository: https://github.com/OceanographyforEveryone/OpenCTD and then navigate to the libraries in the unpacked .zip file.

Squidman32, 2019-10-16:
Thanks Andrew, That got it for me.  After I got the whole library I was
able to place the SoftwareSerial in and it compiled.
Lloyd Trueblood Ph.D.
Associate Professor and Chair
Biology Department

<https://lasierra.edu/>

Lloyd.Trueblood@lasierra.edu
Phone: (951) 785-2 <(951)+785-2105>521
Fax: (951) 785-2100

lasierra.edu  <https://brand.lasierra.edu/email-signatures/#>
4500 Riverwalk Parkway | Riverside, CA 92505

Confidentiality: This e-mail (including any attachments) may contain
confidential, proprietary and privileged information, and unauthorized
disclosure or use is prohibited. If you receive this email in error, please
notify the sender and delete this e-mail from your system.

closed 2020-06-23 by SouthernFriedScientist