OceanographyforEveryone/OpenCTD #23: Raspberry pi instead of Arduino/Adalogger

gshubham96, 2019-11-03:
Hey Guys

I have been interested in this project and have been wanting to build one for quite some time now. I was wondering however if anyone has tried substituting Arduino/Adafruit combo with a raspberry pi. If yes, is there any text out there that I can follow?

Please also let me know if this is not the right place to ask this question.

SouthernFriedScientist, 2019-11-03:
Hello!

Definitely the right place to ask. Steve Haan has a Pi-based OpenCTD/Camera/BRUV set up. I don't know where he keeps his code, but you can find him on twitter and ask: https://twitter.com/HaanHouse

IanTBlack, 2019-11-04:
Steve Haan has a working PiCTD.
Here is his GitHub Repo:  https://github.com/haanhouse/pictd

I've used some of his Python code to get a working version of a sensor
system using the WhiteBox Labs Tentacle HAT for the Pi and a few Atlas
Scientific Sensors.
Here is my Repo: https://github.com/IanTBlack/FlowThrough

If you are looking for a compact solution, the PiZero would be perfect.
Let me know if you run into any issues. I'd be happy to dust off a Pi and
help out.

Cheers,
-------------
Ian Black
Faculty Research Assistant | Ocean Observing Center
College of Earth, Ocean, and Atmospheric Sciences | Oregon State University
Cell: (541) 817-3687

gshubham96, 2019-11-06:
Hey @SouthernFriedScientist  and @IanTBlack.
Thanks for the quick replies!

I did find PiCTD on Hackaday.io. I will definitely check out the links you posted before I begin. I am planning to start building one sometime in late-January 2020 so plenty of time to plan everything before that.

@IanTBlack 
I am hoping to implement this on a custom stack. Not sure how that would work out yet but I will definitely update when I have it figured out. :) Finally, I want to deploy one in a lake. Atlas Scientific seem to have most of the sensors I am interested in. Thanks for posting about it.

Appreciate the help. Considering there might be no big updates till January, should I close the issue or keep it open ?

Regards
Shubham Garg

closed 2020-09-25 by SouthernFriedScientist