OceanographyforEveryone/OpenCTD #25: Independent System Not taking values

MattChace, 2020-06-22:
Uploading the code from my computer Arduino IDE and opening the Serial Monitor I am seeing the sensor take values and store them to the SD Card, I remove the SD Card and read the txt file on my computer and confirm that the values were saved. 
However, when the MCU is powered by a battery and I did a test cast, I open the txt file and find it empty. Not sure why the code is not running when it is turned on by the battery. Any assistance would be greatly appreciated!!

IanTBlack, 2020-06-22:
Hi Matt,
Which version of the OpenCTD are you working with? If you have modified you code, can you attach it?

Thanks,
- Ian

MattChace, 2020-06-22:
Hi Ian!!
Thanks for the quick response! We are working with the Version of the CTD that is within the Construction/Operation Manual and followed all the steps within that text, I have attached my Arduino Code below.

Thanks!

  *   Matt

Matthew Chace

Bachelors of Science in Electrical Engineering

Wentworth Institute of Technology

Boston, Massachusetts

IanTBlack, 2020-06-22:
Hi Matt,
Doesn't look like it attached properly. You might have to do it directly through the GitHub website.

Couple more questions.

1) If you add an empty SD card, is the text file created, but remains empty?

2) Is the SD card seated properly?
   -If you operate it under battery on the bench, you could try tying the EC EZO LED to turn on/off if the SD card is not detected. See lines 140-144 here: https://github.com/IanTBlack/CTDizzle/blob/master/Adafruit_M0_Mk5/2019-04-29_Profile.ino

3) What battery are you using?

Unfortunately I don't have an M0 Adalogger, but I do have a BLE M0 Feather + Adalogger featherwing, so I will see if I can recreate your problem once I get your code.

- Ian

MattChace, 2020-06-22:
My Code is pasted below, I put a new micro sd into the board and turned it on. The MCU doesnt create a txt file at all. 

``//Date and time functions using a DS3231 RTC connected via I2C and Wire lib
#include <Wire.h>
#include "RTClib.h"

//serial peripheral interface and library for SD card reader
#include <SPI.h> //serial peripheral interface for SD card reader
#include <SD.h> //library for SD card reader

//communication protocols for temperature sensors
#include <OneWire.h>  
#include <DallasTemperature.h>   

//Software libary for the pressure sensor
#include <MS5803_14.h>

//EC Circuit uses software serial
#include <SoftwareSerial.h>

RTC_DS3231 rtc; //define real-time clock

const int chipSelect = 4; //sets chip select pin for SD card reader

MS_5803 sensor = MS_5803(512); // Define pressure sensor.

OneWire oneWire(6); // Define the OneWire port for temperature.
DallasTemperature sensors(&oneWire); //Define DallasTemperature input based on OneWire.

SoftwareSerial ecSerial(12, 13); // Define the SoftwareSerial port for conductivity.

double pressure_abs; //define absolute pressure variable

//Declare global temperature variables.
float tempA;
float tempB;
float tempC;

//Declare global variables for eletrical conductivity
float EC_float = 0;  
char EC_data[48]; // A 48 byte character array to hold incoming data from the conductivity circuit. 
char *EC; // Character pointer for string parsing.
byte received_from_sensor = 0; // How many characters have been received.
byte string_received = 0; // Whether it received a string from the EC circuit.

void setup () {

// comment the following three lines out for final deployment
  #ifndef ESP8266
    while (!Serial);   //for Leonardo/Micro/Zero
 #endif

  Serial.begin(9600);

//Initialize SD card reader
  Serial.print("Initializing SD card...");

  if (!SD.begin(chipSelect)) {
    
    Serial.println("Card failed, or not present");
    return;
    
  }
  
  Serial.println("card initialized.");

  delay(1000); 

  if (! rtc.begin()) {
    
[... 230 lines cut ...]
    }
  
    dataFile.print("  ");
    dataFile.print(pressure_abs);
    dataFile.print("  ");  
    dataFile.print(tempA);
    dataFile.print("  ");
    dataFile.print(tempB);
    dataFile.print("  ");
    dataFile.print(tempC);
    dataFile.print("  "); 
    dataFile.println(EC);
    dataFile.close();    

  }
  
  delay(50);

}

void parse_data() { // Parses data from the EC Circuit.

  EC = strtok(EC_data, ",");                  

}`

SouthernFriedScientist, 2020-06-23:
Hello,

Lines 48, 49, and 50...  

`  #ifndef ESP8266
    while (!Serial);   //for Leonardo/Micro/Zero
 #endif `

...provide diagnostic information during assembly and should be commented out or removed from the code prior to deployment, otherwise the Arduino will look for a serial connection before writing to the SD card.

MattChace, 2020-06-24:
That worked!! Thank you so much!

Matthew Chace

closed 2020-09-25 by SouthernFriedScientist

goldsphininx, 2020-09-25:
House cleaning. ;)