OceanographyforEveryone/OpenCTD #30: Conductivity calibration

gheleguen, 2021-08-09:
Hi everyone, 

I made an open CTD for an antarctic expedition. This one must leave before 1st September. 
All is ok. I have just a problem with conductivity calibration. 
When I send c,0 or other command. I didn't got any response. 
The response of conductivity sensor in fresh water is around 300. Is it µS/cm ? 

The baud range is ok and TX and RXpin also right.

I work for an other job with teensy 3.6 board. I think Arduino IDE work with it's SoftwareSerial librairies. 

My Hardware is EZO board and mini conductivity probe k1.0. 

Excuse my english level. I'm french.

Best regards

Guillaume Leguen

IanTBlack, 2021-08-09:
> [This email originated from outside of OSU. Use caution with links and attachments.]

Hello Guillaume,

The documentation for the EC EZO can be confusing. It's been a while since I have built a CTD so I just went through it again.

When you send a "c,0", you are telling the EC EZO to stop auto sampling when in UART mode. If you have responses turned off with the "OK,0" command, then you won't get a string "OK" at the end of each response. To turn on responses, enter "OK,1".

Did you try sending a "R" command after turning off auto sampling? That should return a reading. To turn on auto sampling when you are done with the calibration, enter "c,1".

Given the accuracy of the sensor, I would say 300 uS/cm is reasonable for fresh water. Did you try it in distilled water?

Cheers,
- Ian

gheleguen, 2021-08-09:
Hi Ian,

Really thanks for your queekly response. 
It's working. But in the first time it didn't working. I don't know why. 
But my CTD is calibrated. 

It's bizare but it's ok ;-)

Have a good day

Guillaume

IanTBlack, 2021-08-09:
That's great. I'm glad it is working. Sometimes it is as simple as restarting the whole thing!

I am interested to see how you use the CTD in Antarctica.  Keep us updated!

SouthernFriedScientist, 2021-08-10:
Hello,

Glad to hear you got it working. I've also occasionally had issues with the serial connection while calibrating the conductivity probe, but find a reset will get it moving again. The good news is you shouldn't have to calibrate it very often. 

It is  µS/cm. 

Make sure you reenabled autosampling before re-flashing the OpenCTD software (and that you edited out the three lines of diagnostic code that pause the sensors until a serial connection is established). 

All the best and looking forward to seeing what you do in Antarctica!