OceanographyforEveryone/OpenCTD #37: Problem in RTCLib

pwnel, 2024-02-11:
Code doesn't compile due to problem in RTCLib.  Happens for both OpenCTD_mo.ino and RTC_Reset.ino 

In file included from /Users/pwnel/Downloads/OpenCTD_m0/OpenCTD_m0.ino:3:
/Users/pwnel/Documents/Arduino/libraries/RTClib/src/RTClib.h:25:10: fatal error: Adafruit_I2CDevice.h: No such file or directory
   25 | #include <Adafruit_I2CDevice.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

pwnel, 2024-02-11:
This seems to be solved by explicitly installing the Library "Adafruit_BusIO" in addition.  This is not stated in the build manual.

DrAndrewThaler, 2024-02-12:
Hello and thank you for bringing this to my attention! I'm glad you found a work-around, which I'll make a note of in the readme while we incorporate it into the manual.

closed 2024-02-12 by DrAndrewThaler