OceanographyforEveryone/OpenCTD #40: Adafruit Feather M0 Disconnects during OpenCTD_mo.ino Upload?

EllisWebb, 2025-02-15:
Hello, 
Control unit assembled; IDE, Boards, and Libraries downloaded!

Notes: 

- Arduino IDE 2.3.4
- Mac OS Sequoia 15.2
- Carrier Board 7c
- Adafruit Feather RP2040 AdaLogger Flash with microSD Card (Part No 5980) 

**The OpenCTD_mo.ino compiles without issue seemingly. Unfortunately, as soon as the IDE gets to the upload phase the lower right hand corner of the IDE changes from:**

"Adafruit Feather M0 (SAMD21) on /dev/cu.usbmodem1101"

**to:**

"Adafruit Feather M0 (SAMD21) on /dev/cu.usbmodem1101 [not connected]"

**The output window then throws the following error:**

"No device found on cu.usbmodem1101
Failed uploading: uploading error: exit status 1"

I believe I have ensured I am using the right board, "Adafruit Feather M0 (SAMD21)" and connected to the right port /dev/cu.usbmodem1101 initially, however, during the upload phase it somehow causes the AdaLogger to disconnect? 

**Upon online research there may be a few issues I believe I could be encountering.** 

1. Drivers not automatically downloading? 
_I see no indication of 6.7.3 "Connect your Adalogger to your laptop using a micro-USB cable. The
necessary drivers should automatically install." being successfully completed. Though I also don't know where to look for said drivers..._

2. UBS C - USB C chord is charging only? 
_The Adafruit Feather M0 we are working with has a USB C port not micro-USB. Some websites have recommended to purchase specific USB C chord, this is the first I am hearing of this issue, in any capacity, then again I haven't used many USB C - C chords to date._

3. Other recommendations seem to site AdaLogger Reset protocols or Manual Bootloader protocols? 
_Being novice, we are slightly hesitant to take these steps. _

4. Other issues I haven't listed or read about yet?

**Any advice welcome!** 

Thanks, 
Carter & Rich

IanTBlack, 2025-02-15:
Does the board take up a serial port when you connect via IDE?
Are you able to upload an example sketch from Arduino like blink.ino?
If you are not able to do that, my first thought is that your cable may not be a data cable. But if you are able to see the connection appear in your device manager then maybe it is not the cable.

Did you add the Adafruit URL to the additional boards manager? 
https://learn.adafruit.com/adafruit-feather-m0-basic-proto/setup

IanTBlack, 2025-02-15:
@EllisWebb I just looked over your post again and I just realized you are using the Adafruit RP2040 Feather, not an M0.

You should follow this guide to set it up instead. 
https://learn.adafruit.com/adafruit-feather-rp2040-pico/arduino-ide-setup?gad_source=1&gclid=EAIaIQobChMIrenNxOXEiwMV_RitBh0ydzmwEAAYASAAEgIVTvD_BwE

IanTBlack, 2025-02-15:
Even though you do not have an M0, I suspect that you could get things to run on the RP2040 with Arduino. 
Based on the RP2040 Arduino guide I linked above, it looks like the pinouts do not align with what is on the board. You may need to adjust the wiring or the identified pins in the sketch to get things to work for the OpenCTD. Something to keep in mind.

EllisWebb, 2025-02-17:
This makes a lot of sense, simple build list error.  Thanks for the links. 

Will be an exciting to learn how to change the identified pins in the sketch! 

Thanks for the heads up and the advice,
Carter

closed 2025-02-17 by EllisWebb